sinara-hw/Booster #367: P3V3 rail is programmed for 3.44V

ryan-summers, 2020-08-19:
The P3V3 rail is actually ~3.44V as opposed to the expected 3.3V logic level.

This plays a key role when translating analog samples into real-world voltages. The references in the schematic should be updated to indicate the real voltage of the rail or the resistor divider should be updated to correctly output 3.3V

hartytp, 2020-08-19:
@ryan-summers thanks for all the good work on this.

Reminding myself, the 3V3 rail is currently supplied by the SMPS IC12
[image]

With the current resistor values (chosen to be E24 I assume) and a nominal Vref of 0.8V, we expect 3.434 which is consistent with your observations.

:+1: to adding a note on the schematic.

hartytp, 2020-08-19:
To calibrate expectations here, the critical functionality for this ADC is measuring the PA FET current. The [InAmp](https://www.analog.com/media/en/technical-documentation/data-sheets/LT6106.pdf) offset voltage is 150uV typical/350uV max.

50mA*0.1Ohm gives 5mVsense at the InAmp. That's a 3% accuracy (ignoring any other source of error).

Having an ADC Vref of 3.4 is a 4% inaccuracy. So, it doesn't make a significant difference to the overall current sense accuracy (in any case, biasing the FETs to a constant gate voltage is a slightly crude way of doing things and I'd expect the variations in FET current due to temperature to swamp the inaccuracies in the initial setting).

tl;dr I think this is worth an annotation, but not a change to the PCB

ryan-summers, 2020-08-19:
> tl;dr I think this is worth an annotation, but not a change to the PCB
Agreed.

I just want to point out I'm not trying to use this as a precision measurement device, I'm just trying to show the most accurate information possible to the user

closed 2022-02-07 by gkasprow