sinara-hw/Booster #368: enable/clear interlock transient

jordens, 2020-08-19:
There is about 0.6 dB AM distortion at 33 dBm full AM keying on two different time scales: one on a few ms and another on a few 10 ms timescale.
Is this expected?

[image]

(Excluded attenuator, signal source, Booster input RF switch).

hartytp, 2020-08-19:
I did some testing with a diode detector some time back. see e.g. https://github.com/sinara-hw/Booster/issues/35#issuecomment-413263136

IIRC that's pretty similar to the level of transients I saw with some of the MCL amps I tested and wasn't overly concerning to us

dnadlinger, 2020-08-19:
@hartytp: Just out of curiosity, your measurements probably still would have been with the higher bias currents (~200 mA?) back then, right? Perhaps suggests that the distortion is from some intermediate stage.

jordens, 2020-08-19:
If I did the math right 0.6 dB is quite a bit worse than the ~3 % you measured there.

hartytp, 2020-08-19:
> If I did the math right 0.6 dB is quite a bit worse than the ~3 % you measured there.

Yeah, it is a factor of a few worse.

Either one of us made a measurement error or a change in the design has made the transients worse (e.g. lower bias currents).

Due to the assorted reliability issues we haven't had enough hardware around for testing for a while. We just received a shipment of hardware from CTI so we should now have plenty spare for testing. I can have another look at this next time I'm in the lab (no fixed ETA yet).

hartytp, 2020-08-19:
> Just out of curiosity, your measurements probably still would have been with the higher bias currents (~200 mA?) back then, right?

Yep, it was the higher bias current, also no TVS diodes (although, the timescale for this feels more thermal than diode related).

> Perhaps suggests that the distortion is from some intermediate stage.

I would have thought the opposite...if increasing the FET current reduces the transients then it suggests that the issue is with the PA stage, doesn't it?

dnadlinger, 2020-08-19:
> I would have thought the opposite...if increasing the FET current reduces the transients then it suggests that the issue is with the PA stage, doesn't it?

Yes, of course – I was blindly trusting your statement that the size of the transients was comparable. ;)

hartytp, 2020-08-19:
> I was blindly interpreting your statement as the size of the transients being comparable. ;)

I meant that the transients I measured previously were comparable to/lower than the transients I'd seen in MCL amps, not that the transients I saw were the same as Robert's measurements.

hartytp, 2021-01-11:
@gkasprow when do you think you can have a look at this? We should:
- confirm that these transients are present in the current hardware release (i.e. repeat the measurement for good measure, although I fully expect to see the same effect reported above)
- check which part of the circuit the transients come from (presumably the power stage)
- try to understand the source of the effect. Is it something easy to fix like a capacitor/choke being too small? Or, is this a thermal transient which we can't really do anything about?

I have seen similar effects in MCL amplifiers and I do expect to see something qualitatively like this in any PA. I can't remember off the top of my head the size of the effects I saw when I looked at MCL amps, but I believe they were the same order of magnitude.

We should definitely understand this better and check that there are no easy ways of improving the design (e.g. bulking up a choke/capacitor) but if it turns out that this effect is intrinsic to the current PA design then it's not a disaster as far as I'm concerned -- the design was optimized more for power efficiency than linearity, but if other people have other needs then alternate PA designs for other use-cases are very welcome!

hartytp, 2021-01-11:
It's also worth increasing the bias current and repeating the measurement (I assume the above was taken at 50mA quiescent). I expect that the bias current may give users some freedom to choose between linearity and power efficiency but we haven't played with this yet. 50mA is a pretty low bias current and the amps can take quite a bit more according to the data sheet.

gkasprow, 2021-04-18:
Let me understand it correctly. It happens when the power is suddenly decreased, for example during AM keying.
So if I adjust my RF generator (SMB100A) to work with internal AM modulation, I should observe such effect.
This can be the effect of a fairly slow 28V regulator response. There is some room for tuning.

hartytp, 2021-04-19:
Try using a fast switch (e.g. a mini-circuits one) to turn the RF on and off at say 10Hz. I think this is either a supply issue or a thermal issue as it's largely on slow timescales.

gkasprow, 2021-06-22:
I recreated the issue. The RF switch is [MIKROE-4168](https://www.mikroe.com/rf-switch-click) Click board equipped with [MASWSS0115](http://download.mikroe.com/documents/datasheets/maswss0115.pdf) rf switch.
[image]

gkasprow, 2021-06-22:
longer time scale. Green trace is switch control signal
[image]

gkasprow, 2021-06-22:
I will repeat it with a higher bias current

michalgaska, 2022-01-24:
I recreated this issue. I connected a lab bench power supply to VCC_PWRAMP (P29V0_CH) and I slightly increased the voltage by 1V. The RF signal looks a little bit better. 
[image]

gkasprow, 2022-02-07:
This looks a bit like the setting time of the LDO. I'm not sure if we can do anything with it. A radical solution would be to use better HV LDO. Is it work exploring further?

jordens, 2022-02-08:
The problem may be that Booster is not designed to be a pulse amplifier at all. From the non-linearity of the Vgs-Ids curve that could also be trivially expected. If this is intrinsic to the change and subsequent settling of the average bias current then it's just another instance of distortion (very large AM-AM with inconveniently long time constants).

gkasprow, 2022-02-08:
That should be not the case with HL version.it is very linear and load change do not produce such changes on power supply current.