sinara-hw/Booster #370: interlock flop connection

jordens, 2020-08-20:
Currently, the comparators are connected to the clock inputs while the `on_offn` signal is connected to the asynchronous active low clear input. That's not a good design. The interlocks should **always** (i.e. asynchronously and level sensitive) assert and block the rf switch while they should only ever be cleared on falling `on_offn` (i.e. negative edge sensitive). With the current design, if the interlock thresholds remain asserted when `on_offn` rises again (e.g. because someone connected two amplifier outputs via a splitter and there is some mismatch) you'd never trip the reverse power interlock. It also prevents safe testing of either the interlock thresholds (i.e. disable the switches, set the thresholds very low and hope for them to be triggered). It's not a huge risk but it's clearly wrong.

Introduced in https://github.com/sinara-hw/Booster/issues/192

Suggest to make the comparators an asynchronous set, and the `on_offn` a negative edge sensitive clear.

closed 2020-08-20 by jordens

hartytp, 2020-08-20:
> With the current design, if the interlock thresholds remain asserted when on_offn rises again (e.g. because someone connected two amplifier outputs via a splitter and there is some mismatch) you'd never trip the reverse power interlock

Unless I'm misremembering things, the reverse power interlock is implemented in software only so this isn't an issue.

hartytp, 2020-08-20:
(but I can see it being a pain for debugging etc and I'm not suggesting that the current design is great).

hartytp, 2020-08-20:
The nice thing about these boards is that they're nice and cheap (2 layers, cheap silicon). So it's easy to produce a couple at low cost to test out ideas for better interlocks using an existing Booster before committing to a larger production run.

jordens, 2020-08-20:
Nope. Reverse power interlock is implemented in hardware. See the other issue.
IME, low board price is dwarfed by the hours of debugging/maintaining saved with a clean design, good review, and thorough tests.

hartytp, 2020-08-20:
> IME, low board price is dwarfed by the hours of debugging/maintaining saved with a clean design, good review, and thorough tests.

Sure. Anyway, the point is just that we should prototype any changes before rolling out to users, which is presumably uncontroversial...

jordens, 2020-08-20:
That is indeed uncontroversial.