sinara-hw/Booster #373: v1.5 testing

hartytp, 2020-10-07:
Now that the design is 🤞 pretty stable it's worth repeating some of the characterization measurements previously made.

Things on my list (feel free to add suggestions) are:
- [ ] gain/harmonics v input power at a fixed RF frequency of 200MHz (cf https://github.com/sinara-hw/Booster/issues/2#issuecomment-384684063)
- [ ] large-signal cross-talk
- [ ] large-signal phase noise
- [ ] switching transients on various timescales at high/low output powers
- [ ] spectral purity

hartytp, 2020-10-07:
- [ ] play with the new quartiq firmware

jordens, 2020-10-07:
Note that [this](https://github.com/sinara-hw/Booster/issues/2#issuecomment-384910864) is not how you'd usually measure or define (I)IP3. I have a hard time figuring out what your measurement is saying. Take IP3 as power dependent and compute it directly for each point without fitting. Also for IP3 it's better to take two tones to simplify the frequency dependence.

hartytp, 2020-10-07:
@jordens I agree, that's not how to measure IP3. I was planning to just measure the harmonics as a function of power for now. It's an easy measurement to make with the kit I have to hand and gives a lot of useful information.

I can potentially measure two-tone distortion products, if someone has a concrete measurement specification in mind, depends what kit I can find to hand.

hartytp, 2020-10-07:
Harmonics:

Using a synth and a spectrum analyzer to measure the harmonic content on Booster's output. Working at a fixed frequency of 200MHz. First plot is the synth directly into the analyzer. Second is the synth into Booster into the analyzer via a 30dB attenuator. The second plot cuts out at high powers due to the interlock (didn't have the Booster on the network to change the interlock settings).

[image]

[image]

For reference, I previously measured 
[image] in https://github.com/sinara-hw/Booster/issues/2#issuecomment-384684063 which is broadly comparable

hartytp, 2020-10-07:
Note the second harmonic is only ~11dB down at higher powers, which is quite noticeable on a scope. I haven't checked in detail, but suspect that it's due to the PA stage (although, it's interesting to note that reducing the bias current a long way doesn't seem to have made a difference). I suspect that this is the trade-off one gets for the relatively low power consumption, but don't have well enough calibrated expectations to say much more. In any case, it's acceptable for our anticipated uses...

jordens, 2020-10-08:
You say

>  The AOMs I will drive with it are far more non-linear than Booster, that doesn't concern me.

How non-linear are yours? Last time I looked at AOM IMD (which is admittedly a while ago) I seem to remember some something like 40 dB dynamic range (at typical useful diffraction efficiencies and powers), which is significantly better than Booster (30 dB at 30 dBm).

hartytp, 2020-10-09:
> How non-linear are yours? Last time I looked at AOM IMD (which is admittedly a while ago) I seem to remember some something like 40 dB dynamic range (at typical useful diffraction efficiencies and powers), which is significantly better than Booster (30 dB at 30 dBm).

That comment is from quite a while back and I'm not sure I'd still agree with my past self.

There are a couple of questions I'm interested in here:
- where is the non-linearity coming from? I assume it's dominated by the power FET, but it's worth checking
- are there better designs that offer significantly improved linearity with comparable power efficiency? Or, is this just the trade-off one has to accept if one wants to meet our initial design spec for efficiency?

For a lot of our use-cases this non-linearity is non problematic and the low power consumption is really nice (in some of our labs we're running up hard against the AC limits). Booster was designed to be modular, so people are welcome to design other amp circuits for different specs (IME the hard part with these projects is getting the mechanics/thermal management/power/firmware etc, once that's all done the RF part is a relatively fun and simple project)

hartytp, 2020-10-09:
> where is the non-linearity coming from? I assume it's dominated by the power FET, but it's worth checking

CTI have agreed to measure the harmonic content at a few points in the signal chain to verify that the distortion is dominated by the output stage. They'll do this in 3-4 weeks when they assemble the next batch of Boosters

gkasprow, 2020-10-09:
We can always make a module version that offers much better linearity but with poor power efficiency. We have a lot of margin in cooling capabilities.

gkasprow, 2020-10-09:
For your applications that require high linearity, do you need such wide bandiwth ? I'm using very nice low cos [modules ](https://www.mitsubishielectric.com/semiconductors/content/product/highfrequency/siliconrf/module/ra30h3847m1.pdf)that offers 30W of power. Linearity is good, it is able to broadcast 8k OFDM carriers at 10W without issues. But it is narrowband, works in ~350...500 MHz region.

hartytp, 2020-10-11:
> For your applications that require high linearity, do you need such wide bandiwth ? I'm using very nice low cos modules that offer 30W of power. Linearity is good, it is able to broadcast 8k OFDM carriers at 10W without issues. But it is narrowband, works in ~350...500 MHz region.

To be clear: right now I'm not aware of an application that requires better linearity. We shouldn't start designing anything until we have a concrete specification.

My point was just that the current Booster design works well for us, but if someone wants something different then it's relatively easy to redesign the RF stage. But, let's wait until someone actually asks for that...

In any case, I will be curious to see how the linearity looks at different points in the signal chain. I suspect that the distortion is dominated by the output stage but if not we should consider design improvements...

gkasprow, 2020-10-11:
OK, I see your point:)

hartytp, 2020-10-16:
Harmonics over the full output power range (i.e. repeat of the previous with the interlock set to +38dBm):

[image]

hartytp, 2020-10-16:
And, here is the deviation of the measured power in the fundamental from the gain fitted at low powers

[image]

hartytp, 2020-10-16:
Looking for spurs...

[image]
1MHz span/1kHz BW

100kHz/100Hz
[image]

hartytp, 2020-10-16:
Note that the source isn't amazingly low noise and the analyzer isn't a fancy phase noise meter.

Now repeating the above measurements with the synth directly into the analyzer (no 30dB attenuator).

[image]
10kHz/100Hz

[image]
1kHz/100Hz

hartytp, 2020-10-16:
Conclusion is that the spectral purity of Booster is as good as I can measure with this synth/analyzer combo (I expect the limitation is the 8656B). So, it passes the "it's as good as a medium quality synth" test. Next time I have access to it, I'll put Booster on our fancy phase noise meter, but I'm already quite happy with this level of performance...

closed 2021-04-18 by hartytp