sinara-hw/Booster #375: Interlock tripping too early

hartytp, 2021-01-05:
@gkasprow I'm seeing an issue with the interlock tripping at too low a power, which I do not understand. 
- Using the latest Booster HW rev with latest quartiq firmware
- Drive Booster from a synth (HP8656B) output 50Ohm terminated, frequency 100MHz
- For a variety of interlock set points I tested between 13dBm and 33dBm the interlock reliably trips ~6dB 
- the internal + internal power detector readings closely match measurements using an external power detector
- after calibrating, I find the gain + offset for the two detectors closely matches the expectations from the component data sheets
- for 33dBm interlock set point, I probed the output power Vref (one comparator input) using a 10x scope. This gave 1.368V as expected (around 5mV noise measured, but probably measurement pickup)
- around 1dB below tripping, the power detector output measured 1.132V (again, not much noise there)
- The detector slope is  35mV/dB so that's consistent with the interlock tripping 6dB (210mV) early
- I checked that the synth doesn't have glitches when toggling the power by putting it directly onto a scope in single shot mode and toggling the power between various different values
- I then put a scope on the detector output (comparator input) by probing on the connector of supplemental control PCB
- I then toggle the synth up and down by 1dB and look for glitches at the detector output by using a scope in single shot mode
- If the interlock threshold is more than 6dB above the powers I'm using, I don't see any glitches on the detector
- if I change the interlock threshold to be 6dB (200mV) above the higher power I'm using then toggling the synth does produce a 200mV transient (I'll post a picture later on). So the transient on the detector output is only present when the interlock is set 200mV above.

Any ideas?

hartytp, 2021-01-05:
Here is one observation that I think is relevant:
- the comparator is fast and prone to oscillations (we had to use positive feedback to stabilize it)
- there is an AC 100R : 1k divider between the OpAmp output resistor and the feedback resistor
- there is around 2V difference between the 3V3 comparator supply rail and the detector output voltage

So, we do expect a 200mV glitch on the detector signal whenever the interlock trips. This is not unexpected nor problematic.

However, we don't expect this AC positive feedback to cause the interlock to trip earlier.

Maybe this is a red herring and the problem is elsewhere, but it feels relevant.

Thoughts @gkasprow ?

hartytp, 2021-01-05:
[image]

hartytp, 2021-01-05:
This glitch is pretty fast. 200ns / div (40ns/sub division( looks like a rise time of around 5ns. That's way outside the power detector or opamp bandwidth. The comparator hysteresis is 100pF || 1K so a 100ns rise time.

The glitch I see could also be something like measurement pick up from when the comparator output switches so actually a measurement artefact and not the cause of the interlock tripping.

hartytp, 2021-01-05:
@gkasprow I think that to understand this someone needs to probe the interlock circuit in a few places to try to see what's actually going on. (e.g. is that glitch before or after the comparator switches?)

I can't easily do this because the places one needs to probe are on the underside of the control PCB. So, the control PCB needs to be removed from the RF module and connected with cables.

Are you / one of your students able to investigate?

hartytp, 2021-01-05:
NB I assume the reason this wasn't noticed before was that the old firmware calibrates a separate slope + offset for the interlock threshold and the power readout. In contrast, the new firmware currently uses the same for both.

gkasprow, 2021-01-07:
I have a setup that allows me to investigate the issue with the cover open and upper PCB removed.
I don't have any Booster right now.
@akaminska is there any chance to get one Booster for debugging?

hartytp, 2021-03-18:
@gkasprow could you have a look at this issue next please? It would be good to understand this before more Boosters are manufactured.

gkasprow, 2021-04-18:
@hartytp can I investigate it with old firmware? Did it happen before switching to new firmware?

hartytp, 2021-04-19:
I haven't tried with the old firmware, but I would expect so.

The observation is that the comparator reliably switches when the two inputs are not equal. I checked for glitches that could cause the switching but didn't see anything.

Try setting an interlock level and monitoring the interlock DAC + power detector output while increasing the input RF power until the interlock trips.

gkasprow, 2021-05-03:
We can also think about using two detectors running in parallel + two amplifiers and two Stabilizer channels. Then we can write a correlation-based algorithm that gets rid of shot and flicker noise.

gkasprow, 2021-06-23:
@hartytp indeed, the interlock trips when there is ~200mV difference between detector output and DAC output.
The DAC output is 200mV higher than the detector output.
yellow: RF out after 20dB attenuator
green - RF switch control signal
blue - DAC output
violet - detector output
[image]

gkasprow, 2021-06-23:
It looks like your glitches come from another source because I don't see any, but there is an effect you describe!

gkasprow, 2021-06-23:
As you wrote, old firmware calibrated the detector and interlock independently so we didn't notice the issue.

gkasprow, 2021-06-23:
I added the comparator output signal. Nothing suspicious here
[image]

gkasprow, 2021-06-23:
I looked at the DAC signal. And just before the transition, it is 200mV lower than the detector output!
[image]

gkasprow, 2021-06-23:
So, the question is what causes this 200mV pulse at the DAC output...

gkasprow, 2021-06-23:
I got it! DAC produces 200mV glitches every 10us!

gkasprow, 2021-06-23:
It seems it picks the glitches from the SCL line - they are routed close together!
[image]

cjbe, 2021-06-23:
Oooooff! Good catch!

gkasprow, 2021-06-23:
There are four solutions:
- stop input power ADC I2C activity which also means no input power interlock
- calibrate the interlock and power detector separately as in old firmware
- add a 100nF capacitor at the comparator input. Can be done at any time after production
- fix the trace on the PCB

gkasprow, 2021-06-23:
1nF capacitor between pin 4 and 5 of the comparator solves the issue

gkasprow, 2021-06-24:
[image]

gkasprow, 2021-08-04:
It was fixed in v1.5.1 of the board.

closed 2021-08-04 by gkasprow