sinara-hw/Booster #377: project status

hartytp, 2021-01-11:
Project status update for anyone following:

Since fitting the TVS didoes, all the reliability issues seem to be resolved (statistics: >50 channels for >6 months with no reported issues or other glitches --- other than one channel which it seems didn't get patched correctly due to a communications error).

The new firmware developed by Quartiq now works nicely. We're working out a few minor issues at the moment, and testing has been limited so far (that should change over the coming months) but it's already superior to the previous version and seems to work quite nicely (thanks!).

There are a couple of issues that need investigation (see below) but, all in all, I think this is now production ready.

1. There is an outstanding issue with the interlock that needs to be looked at: https://github.com/sinara-hw/Booster/issues/375 (@gkasprow when do you think you can look at this?) but a simple software work-around seems to work
2. We should look into [these](https://github.com/sinara-hw/Booster/issues/368) transients and try to understand their source slightly better.

gkasprow, 2021-01-11:
I have no Booster to look at it. CTI is producing a batch of them. I reserved one of them for debugging purposes.

hartytp, 2021-04-18:
> The new firmware developed by Quartiq now works nicely. We're working out a few minor issues at the moment, and testing has been limited so far (that should change over the coming months) but it's already superior to the previous version and seems to work quite nicely (thanks!).

Current status: the device panics regularly (~once per day or so) when left idle due to https://github.com/sinara-hw/Booster/issues/383 so still not at a point that I would consider production ready