sinara-hw/Booster #378: Cost/robustness/support/maintainability improvements

jordens, 2021-01-25:
From going over the design a couple times now and having been mostly not involved in the initial ~5 revisions, there does seem to be a lot of costly (design, parts, and firmware) legacy and there are significant chances to reduce part/board/mechanics/assembly/software issues. These aren't really major issues individually. But in sum addressing these would make Booster a fully convincing mature design to me.

* A massive cleanup of the schematics is in order. It's really hard to read, the signal flow, arrangement, naming etc is all over the place. This is a major pain point and cost factor for firmware development, debugging and maintenance.
* Parts count with lots of legacy: There are 2 different types of I2C DACs. There are 3 different types of ADCs being used including the CPU ADC and two types of I2C ADCs.
* Having that many components on the main board makes devices with reduced channel numbers needlessly expensive (lots of unused hardware). A device with two channels is about half the price of the 8 channel device.
* The channel supply and control board doesn't do much. For some reason large parts of its functionality (supply and control) are down on the channel RF board or up on the main board. I would look at eliminating the supply/control board in a new revision. Instead either have a shielded pocket/strip in the channel module for non-RF (and move as much as possible from the main board in there) or move everything to the main board. That would reduce the mechanics and could remove an entire board (plus connectors, mounting, assembly/disassembly).
* For me a desirable interface between the RF modules and the main board would be reduced to: P6V5, N8V0, P30V0, I2C. The rest should be in the module. This would also align much better with the possibility to design different amplifier bricks. It's unclear to me what the design requirements for the current interface were.
* The internal comparator and flip-flop of the power detector should be used to protect against fast transients. That would remove a large part of discrete logic. In my experience fast transients are common and really dangerous for downstream parts (within their acceptance bandwidth of dozens of MHz) and I suspect that the current slow interlock will not protect them from a significant part of the existing turn-on-transients. This can't be solved in software. But a slow "thermal" interlock could just as well be implemented on the CPU. The proper approach would appear to be to have both.
* As indicated elsewhere, the flip-flop logic (clock, reset) is badly designed and can render the interlock completely ineffective if not being worked around in firmware.
* Go for more sheet metal instead of custom milled aluminum parts.
* Mechanically the assembly is borderline impossible. The D15 connectors need straight insertion but that's prevented by the SMAs requiring straight insertion through the panel on the other axis.
* I've seen the power supply monitor indicate power supply issues. Either the monitor is not working or the supplies are wonky.

hartytp, 2021-01-25:
> I've seen the power supply monitor indicate power supply issues. Either the monitor is not working or the supplies are wonky.

While we won't know for sure until the issue is resolved, I'd put my money heavily on the monitor. We've had quite a few issues with that ADC (and it's currently not supplied correctly) whereas I have probed that LDO carefully and not seen anything suspicious.

hartytp, 2021-01-25:
> But in sum addressing these would make Booster a fully convincing mature design to me.

It's pretty clear that there are plenty of ways this design could be cleaned up and improved (although that's the case with most designs). If we were having this discussion at an earlier stage in the design then I would be enthusiastic about most if not all of your suggestions. However, given where we are, it's not clear to me whether the benefits these changes bring are sufficient to justify the time that would be spent implementing, prototyping, testing, modifying firmware, supporting multiple hardware revisions, etc.

hartytp, 2021-01-25:
Other than the mechanics, I don't think cost is a good reason for these changes. How much would the BOM simplifications actually save per Booster? How much money would be spent on the time to implement them?

hartytp, 2021-01-25:
> Having that many components on the main board makes devices with reduced channel numbers needlessly expensive (lots of unused hardware). A device with two channels is about half the price of the 8 channel device.

I might be wrong, but last time I looked at the cost breakdown from TS, the extra components you mentioned on the main board did not actually make that big a difference to the cost. The biggest driver is batch size.

hartytp, 2021-01-25:
> The channel supply and control board doesn't do much. For some reason large parts of its functionality (supply and control) are down on the channel RF board or up on the main board. I would look at [...]

This PCB is basically just a very low cost wiring harness to get signals from the main PCB to the RF PCB. We've discussed this a few times and the steer from @gkasprow was that eliminating this would drive the price up since basically any other means of routing the signals is more expensive. Maybe with a major overhaul of the mechanics we could eliminate this, but that's not totally clear.

hartytp, 2021-01-25:
> The internal comparator and flip-flop of the power detector should be used to protect against fast transients. That would remove a large part of discrete logic. In my experience fast transients are [...]

I don't agree here. The internal comparator is (a) quite annoying to drive since it needs a lot of current so we'd likely need an additional negative rail to drive it (b) it triggers off very fast transients which are commonly produced by switches etc. At least for the situations I've used Booster in, these are not harmful.

I don't think it's fair to characterise this as removing a large piece of discrete logic: a comparator and a DFF. 

Which turn on transients are you referring to? I'm not aware of Booster producing any major turn on transients these days (since we filtered the interlock switch control signals). Certainly nothing that's harmful to any of the loads we wish to drive.

hartytp, 2021-01-25:
The interlock could potentially have been done in software, but it's not clear to me that would be a better solution

hartytp, 2021-01-25:
> As indicated elsewhere, the flip-flop logic (clock, reset) is badly designed and can render the interlock completely ineffective if not being worked around in firmware.

Yep, that could have been done better

hartytp, 2021-01-25:
> Mechanically the assembly is borderline impossible. The D15 connectors need straight insertion but that's prevented by the SMAs requiring straight insertion through the panel on the other axis.

I've assembled a few of these units and haven't found it that bad with care. Improvements welcome though.

hartytp, 2021-01-25:
> Go for more sheet metal instead of custom milled aluminum parts.

I think we discussed this with @gkasprow some time back and came to the conclusion that the metal enclosures aren't particularly more expensive than alternatives (but I'm happy to be corrected on that).

Having said that, the mechanics do seem to be overkill at the moment. e.g. I don't think that we actually need the fans at all given the low power consumption and decent conductive cooling to the case. I'm not sure if the thinking was to leave them there in case someone wanted (e.g.) to design a high linearity amp which would dissipate significantly more power, or if it's just a question of over-engineering the original design. Even if it doesn't affect the cost too much, shaving some weight (and maybe even some height) off would be nice.

gkasprow, 2021-06-05:
I'm thinking about migrating Booster to an EEM form factor. I mean the Booster will be produced until there is demand for it, but it's time to re-think the approach as @jordens pointed out. It's a long-term project so I'm just collecting feedback.
The [SILPA](https://github.com/sinara-hw/SiLPA) approach seems to be the right one. It can live as a single channel, can be mixed with other EEM modules in case you need just one or two channels, it also uses low-cost mechanics.
What if we move Booster to the same platform?
The CPU board could be made as an EEM module, connected with other amplifier boards using either EEM ribbons ( slight overkill) or MODBUS protocol. Once we de-centralize the control, the main module would only gather the diagnostics.
We can also re-use the existing architecture and connect just the I2C interface.
I don't like I2C because there are always some issues with it. I prefer a serial port (UART, Modbus) or SPI.
In the SILPA design, I added a MODBUS connector to be able to combine several such modules together. I also implemented a full EEM connector (SPI over LVDS, I2C)
Of course, one can use Kasli for that purpose, but that would be overkill.

For the moment I see 3 approaches to modular power amplifiers.

1. Use the existing EEM form factor with EEM connector (SPI slave, I2C, 12V)
- it has the ability to cooperate with other EEM modules, one can plug one Urukul and two 2-channel EEM PAs into the same subrack and save a lot of space.
- can work stand-alone, without supervision, just connect 12V supply. One can use a USB hub to supervise the modules.
- Kasli is expensive if one wants to build an 8-channel Booster equivalent.

2. Use the existing EEM form factor with EEM connector ( I2C, 12V)
- build a dedicated controller that replaces Kasli - essentially take Booster mainboard + IFC board, remove the channel sections
- keep the fan controller, connect it to the COTS 12V fan tray. The ADCs would be accessible either via I2C or MODBUS (over LVDS)
- route all modules using I2C and EEM ribbon cables. One can also implement SPI over LVDS 

3. Make modules with just MODBUS, connect just 12V, and use Ethernet to MODBUS converter when somebody needs remote control. One can also use one of the modules as a master connected over USB to the PC and communicating with other modules using MODBUS.
- it's the lowest cost approach but not very elegant.
- it would require two dedicated firmware versions for the module.

The SiLPA design already has an EEM connector with I2C and SPI over LVDS, it also has MODBUS + address select switch so can be used in all three use cases. We can always DNP unused interfaces and produce the modules with the only I2C to keep the existing Booster architecture. The idea is to get rid of numerous ADCs and DACs and replace them all with low-cost STM32 MCU that has both ADCs and DACs.

For the moment we have 3 versions of the RF modules
- Standard one used in the Booster. In the case of EEM, we can use the RFPA3800 power stage which uses a single 7V and outputs 5W. I ordered the devkit and will play with it within another project.
- High Linearity one, delivering up to 34dBm of power. it can be build using a 12V version of the power doubler.
- High Linearity, very wide BW one, delivering up to 30dBm of power.

The EEM modules can be optimized in such a way that the aluminum shield is very low cost. The essential cost is related to the number of steps that cannot be automated. If the entire part can be made within one process (attach the metal block and mill all areas with a single run), the cost is very low. However, if one needs to precisely mill another side, it needs manual positioning. If one needs to cut holes from the side, it adds a lot of costs. Recently I designed a[ heatsink for 8 channel HV](https://github.com/sinara-hw/HV_AMP_8CH/issues/4#issue-912244423) amplifier and was surprised by how quick cheap it was. It was cheaper than the COTS heatsinks from AAVID that I wanted to use for SiLPA. So, we can make a two-part clamshell heatsink that surrounds the RF board from both sides, providing both heat dissipation and good shielding.
In case the crosstalk is not an issue, one can use the module without the shield at all.

jordens, 2021-06-05:
All agreed.

* I'm a bit unsure regarding the actual demand for the high linearity amplifiers in the AOM range of frequencies. As long as there is no hard demand and hard spec for IMD, maybe we should postpone that. I checked with some of users who complained about the Booster distortion but couldn't get an actual synthesized requirement.
* The simplicity and efficiency aspect is in very high demand. IMD3 (but not e.g. the µs-ms scale distortion) at the level of Booster does look acceptable if characterized fully and stated clearly. And it helps a lot with the thermal management/efficiency.
* On control and monitoring I also agree that EEM isn't very desirable. Modbus-rtu over rs485 seems OK from scanning through the spec, existing code base (on e.g. stm32 and rust) and tutorials. I agree that it's likely preferable over I2C, SPI, Ethernet, CAN for this.
* Re power and control wiring I really dislike cables for this. Even boring DIN 41612 would be better than reaching into the crate to plug/unplug connectors at the back of the cards.

With that in mind building a platform that can do what Booster can do (or with some deliberate trade-offs) but simpler, in 4 HP, with one or two channels, and optional modbus would be a very valuable device.

gkasprow, 2022-02-07:
we have both standard and HL versions now. We will move towards the Eurocard approach -> SILPA project.

closed 2022-02-07 by gkasprow