sinara-hw/Booster #380: High harmonic content

gkasprow, 2021-01-29:
One of the users reported that Booster has high harmonic content. Here are the measurements at 150 MHz Signal, (-18 dbm). Signal was generated by Urukul
[image]

Comparing with MiniCircuits amp
[image]

The Booster was optimized for efficiency, not for linearity.
Let's see what we can do about it

hartytp, 2021-01-29:
See https://github.com/sinara-hw/Booster/issues/381#issue-796716735 for more context

hartytp, 2021-01-29:
@gkasprow while you're looking at this, can you have a look at the (thermal?) transients @jordens measured as well https://github.com/sinara-hw/Booster/issues/368 The two may be related.

gkasprow, 2021-02-04:
I connected the Booster to 400MHz, -20dBm signal and probed the traces with SA and 1:10 probe at the output of I, II and III stage and what I got:
I  stage 2nd harmonics is -42dB
II stage 2nd harmonics is -25dB
III stage 3rd harmonics is -23dB

So, we have a lot of distortion even for relatively low power

For 400Mhz and -18dBm, I got
I  stage 2nd harmonics is -40.7dB
II stage 2nd harmonics is -22,8dB
III stage 3rd harmonics is -20.4dB

It means that it's the second stage who is guilty.

hartytp, 2021-02-04:
That's interesting. And good as it means we may be able to significantly improve this without compromising the power efficiency...

gkasprow, 2021-02-04:
I removed the output TVS and the 2-nd harmonics is -21.8dBm

hartytp, 2021-02-04:
What does the feedback do to the amp's P1dB?

gkasprow, 2021-02-04:
I've just removed it and it seems it does nothing for 400MHz. 
The 2-nd harmonics is still -21.6dB, the output signal slightly increased

gkasprow, 2021-02-04:
At -40dBm of input power, the II stage has -46dB of 2nd harmonics

hartytp, 2021-02-04:
Are these numbers dBc?

gkasprow, 2021-02-04:
Yes

gkasprow, 2021-02-04:
It looks like the PHA-1+ amplifier is working with higher power than planned...

gkasprow, 2021-02-04:
PHA-1+ amplifier is working in very comfortable conditions.

gkasprow, 2021-02-04:
Let's see how it behaves loaded with just SA...

gkasprow, 2021-02-04:
With -18dBm of input power, it shows -2dBm and 2-nd harmonic is -46dBc

gkasprow, 2021-02-04:
So there is something wrong with the III stage input impedance

gkasprow, 2021-02-04:
I replaced L2 with a 300R resistor as in the DS, and it's the same. Let's see what VNA shows...

gkasprow, 2021-02-04:
the S11 is lower than -10dBm, but of course, that won't show us nonlinearities...

gkasprow, 2021-02-04:
what we can do is to add an attenuator between II and III stage and later on remove one before the II stage. That would isolate the stages...

gkasprow, 2021-02-05:
I inserted a 7dB attenuator between the stages and the second harmonic dropped to -28dBc. After I increased the input signal by 7dB the 2-nd harmonics is still at -21dBc

gkasprow, 2021-02-05:
It looks like we have to go for the higher power amplifier. I identified[ one](https://pl.mouser.com/datasheet/2/412/TQP7M9105_Data_Sheet-1500621.pdf) with identical package, supplied from 5V and capable of delivering +30dBm P1dB. I will order a few pieces and give it a try.

hartytp, 2021-02-08:
@gkasprow out of curiosity: (a) are your measurements consistent with the P1dB of the current amplifier? (b) what does the feedback do here, are we sure it is unconnected to this?

gkasprow, 2021-02-08:
the feedback is supposed to lower the gain slightly and improve linearity. But it is 1k : (50R || 50R) so it's influence is negligible.
The first think I did was removal of the FB loop.
I'm pretty confident that the non-deal load of the power stage somehow interacts with the preamp stage.
I'm not able to observe any distortion using scope. But the non-linearities we are talking are not very high.

hartytp, 2021-02-08:
okay, makes sense. 

For testing, this sounds like a really sensible plan. If we're going to make changes to the RF section it might be worth having a think about simplifying things.

hartytp, 2021-02-08:
@jordens @cjbe what are your feelings about the ideal gain for this amp? I feel that 40dB might be a bit higher than optimum. 35dB (at the centre of the frequency range) would be a better match for much of what I want to do.

gkasprow, 2021-02-08:
the input attenuator was added to protect the input stage against overload

hartytp, 2021-02-08:
For my own interest, here are a few MCL options....

[image]

hartytp, 2021-02-08:
It's a question for all the stakeholders here, but I wonder if we'd be okay with removing all the attenuators and going for a https://www.minicircuits.com/pdfs/TSS-13LN+.pdf as a single-stage amplifier. Cheaper, simpler design with less power consumption. Lower overall gain which I personally don't think is a bad thing. Lower noise figure. A bit less input protection, but should be fine with the diodes.

It would mean we have a bad VSWR with the interlock tripped (due to no input attenuator + reflective switch). But I don't see that as being a problem. If it is, we can always use a non-reflective switch (IIRC the motivation for the reflective switch was the switching transients, but that doesn't seem to have been any better with the reflective switch)

hartytp, 2021-02-08:
> the input attenuator was added to protect the input stage against overload

As a historical statement, I don't think that's quite correct. At least, my memory was that we had too much gain and needed an attenuator and figured that we might as well stick it in the input. I don't think this was a case of "we thought about it carefully and decided that we needed an attenuator there to protect the amp". Particularly with the TVS fitted, is there any reason we should be worried about that?

hartytp, 2021-02-08:
What is annoying is that to get, say, 1W one needs -10dBm input. That's actually -16dBm at the first amplifier. So our overall noise figure is more like 20dB which is pretty poor -- particularly when one considers the fact that we're using an amp with a 2.6dB noise figure.

This was the thing that stuck out in my phase noise measurements: the close in noise was very low (I couldn't measure it) but the broad-band noise was surprisingly high. Not that this is a problem for any applications I have in mind, but it does feel like poor design to me and something that's easy to fix.

My position here is that the NF isn't enough of an issue on its own to warrant changing the design (which would require prototyping + testing). But, if we are going to make changes, so have to go through that anyway, then it does make sense to do a bit of housekeeping while we're at it.

gkasprow, 2021-02-08:
> 

    the input attenuator was added to protect the input stage against overload

As a historical statement, I don't think that's quite correct. At least, my memory was that we had too much gain and needed an attenuator and figured that we might as well stick it in the input. I don't think this was a case of "we thought about it carefully and decided that we needed an attenuator there to protect the amp". Particularly with the TVS fitted, is there any reason we should be worried about that?

I think you are right. Protection of amplifier wasn't the main issue. It was rather the S11 of the amplifier with the switch off, that's why we moved the attenuator. AFAIR the previous switch caused high transients so we changed it to HMC.
With TVS we don't care about the input levels, but we still care about S11.

hartytp, 2021-02-08:
> I think you are right. Protection of amplifier wasn't the main issue. It was rather the S11 of the amplifier with the switch off, that's why we moved the attenuator. AFAIR the previous switch caused high transients so we changed it to HMC.

I'd have to dig through the issues to be sure, but my memory is this:
- the non-reflective switch had transients so we switched to the reflective switch because someone suggested it wouldn't have transients (NB we really should have prototyped this first!)
- the reflective switch had pretty much identical transients so we ended up just slowing down the transients to the point where they didn't cause problems

As a result, I'm not convinced that the non-reflective switch would be any better/worse here. Before changing, someone should test this (can easily be dead bugged at 100MHz).

> With TVS we don't care about the input levels, but we still care about S11.

That is a claim that I'm not 100% sure I agree with. Which user specified this? I do not have good visibility into all the applications people want to use this for (@jordens may know more). But, my thinking is that the interlock tripping is a rare event. So the threshold for caring here is not "poor S11 may make the source behave a bit oddly" but rather "poor S11 may break the source".

The only sources that I am considering (again, others should weight in if they feel differently) are Urukul, phaser and synths. If any of those broke with an open load I would not have much kit left in my lab ;)

hartytp, 2021-02-08:
So, unless someone else has input, I would suggest that we can tolerate a reflective switch here. Even if we can't I would suspect that (although it needs testing) a non-reflective switch would probably work fine with the filtered control lines...

Anyway, this is all getting ahead of ourselves. Let's see if the higher P1dB amp improves the harmonic distortion and then decide what other changes we want to make

hartytp, 2021-02-08:
> This was the thing that stuck out in my phase noise measurements: the close in noise was very low (I couldn't measure it) but the broad-band noise was surprisingly high. Not that this is a problem [...]

Well, maybe "poor design" is a little strong here; Booster was designed for things like Doppler cooling using the AD9910 with its internal PLL. In that configuration the amp NF is not a limitation. However, it feels like we can make something which is better without sacrificing in other areas, so we should do so.

gkasprow, 2021-02-08:
Anyway, we have a directional coupler at the Booster input which should redirect the reflected power to the load (two 100R resistors)...

hartytp, 2021-02-08:
Maybe I'm missing something, but that's not my understanding of how a directional coupler works. They are reciprocal. So the loss in either direction is the same; either way 10% of the power gets directed to one of the coupled ports, but 90% goes through the coupler. i.e. they are not an isolator/circulator

gkasprow, 2021-02-08:
Yes, you are right. I mixed it up with a circulator. It redirects only 10% of the power. But again, we have a switch and TVS that protect the amplifier.

hartytp, 2021-02-08:
Indeed. So as long as the sources can take a high S11 load without damage (which Sinara can) I think this isn't something we should worry about.

jordens, 2021-02-08:
I don't think NF is a particularly interesting goal for a power amplifier. I wouldn't be worried about 20 dB and I don't get why one would use Booster as an LNA. In that case, S11 would be very relevant.
As long as there is no consensus on the target output power of the sources, we can't spec the gain (given fixed P1dB of the output, which is by the way unclear to me).
I'd go for 0 dBm input power to meet P1dB. IIRC with some adaptations on Phaser, Urukul, Mirny, Pounder, we can meet this at the source side, but it needs careful checking and analysis and some experiments+characterization. But this is should be the go/no-go.
We also should ensure and afford that all elements other the final amp are not yet compressing at the final stage P1dB. I think Urukul and Phaser are a bit problematic here. Otherwise this becomes disappointing and extremely hard to understand (see below).
And finally, we need a specification for the distortion. Taking the MCL or "better than currently" as a target doesn't make sense IMO. Assuming the target is a AOM/EOM, then both IP3 and IP2 are universally near 10 dB diffraction efficiency. Given your working point RF power, you'd spec the amplifier OIP3 and OIP2.
For Booster currently, the IMD curves are unfortunately not at all clear, making this yet more difficult. There appear to be multiple IMD sources of comparable magnitude interfering leading to non-trivial compression. I think when measured as the more sensible 2-tone-IMD, and when looking at the stages in isolation, that might look very different again, and would give insight. This also needs to be done.

hartytp, 2021-02-08:
> In that case, S11 would be very relevant.

To be here, the above was specifically S11 with the interlock tripped. Which is, as far as I can tell, not a major consideration in any target applications.

> I don't think NF is a particularly interesting goal for a power amplifier. I wouldn't be worried about 20 dB and I don't get why one would use Booster as an LNA

It's not a major consideration, but no point making it 20dB worse than it needs to be

hartytp, 2021-02-08:
> I'd go for 0 dBm input power to meet P1dB. IIRC with some adaptations on Phaser, Urukul, Mirny, Pounder, we can meet this at the source side, but it needs careful checking and analysis and some experiments+characterization. 

I agree, that 0dBm -> P1dB a sensible target to aim for if the Sinara sources can reliably hit 0dBm. I suspect that will translate into ~35dB of gain or less, which would allow us to simplify the RF path/remove a pre-amp stage.

> We also should ensure and afford that all elements other the final amp are not yet compressing at the final stage P1dB. 
> For Booster currently, the IMD curves are unfortunately not at all clear, making this yet more difficult. There appear to be multiple IMD sources of comparable magnitude interfering leading to non-trivial compression. I think when measured as the more sensible 2-tone-IMD, and when looking at the stages in isolation, that might look very different again, and would give insight. This also needs to be done.

Indeed. The current RF path never got properly characterised because we were too buys fighting fires elsewhere. Not that other things are largely extinguished (modulo https://github.com/sinara-hw/Booster/issues/379) it's time to come back and have a look at things we can improve. 

-----

To check we're all on the same page though: while I think there is some low hanging fruit in terms of things we can improve in the design, the current design is IMHO very much fit for its intended purpose (doppler cooling etc where IMD etc is not an issue). If people want it to do more than that it would be helpful to have a discussion about concrete specifications. Without that , it's going to be a case of "fix the obviously suboptimal parts and leave it at that".

jordens, 2021-02-08:
I don't think we are on the same page. If you want to eat or sell fruit, you need to harvest everything and to properly and thoroughly care for the trees. The joy of only plucking low hanging fruit is quickly replaced by disappointment over rotten fruit and  sprawling untrimmed trees. To me, there is still not enough data on how well it works and for what purposes (you don't use it for gates?). This appears to be (at least, but still only) a minimal viable product. Stopping there and not looking at the complete picture would be a mistake. There are quite a few more things to be done (#378).
I think we do agree though, that without concrete specifications and significant work (on characterization, design, and tests), it's hard to get much further.

gkasprow, 2021-02-18:
I replaced the preamp with TQP7M9105 and the harmonic situation is identical. It makes no sense to investigate it further this way. Let's make the correct simulation model first and see what comes out of it.

hartytp, 2021-02-18:
Interesting! Sounds like a good plan

hartytp, 2021-02-19:
Did you/can you measure the S11 of the final stage amplifier with a large signal? I suspect this may be the power stage presenting the pre-amp with a non-linear load at higher powers.

gkasprow, 2021-02-19:
I did. It was better than 10dB. I'm not sure what the signal level was. I improved the matching of the finals stage by copying the datasheet but it did not help.

gkasprow, 2021-02-21:
I ordered NPA1007. It has a similar package and the same voltage but twice higher power. I will give it a try.

gkasprow, 2021-03-19:
I installed NPA1007 and the situation is nearly the same apart from lower gain.
For all measurements, I use 21,9dBm output power.

Default amplifier configuration
Idd; 2nd harmonic:
50mA; -13,8dBm
100mA; -20,7dBm

With NPA1007:
Idd; 2nd harmonic:
50mA; -13,7dBm
100mA; -21,0 dBm
130mA; -25,1dBm
150mA; -27,3dBm
200mA; -31,8dBm
250mA; -32,7dBm

I cannot set a higher current than 250mA.
But I think I know what's going on.
The reference ZHL-2W amplifier consumes 2A at 28V. This is a lot of current!
If we use 4 NPA1007 working with 250mA running at 1/4 of power, we would get -43dBm of 2-nd harmonic.
So there is an obvious tradeoff: harmonics vs Idd.
We can also go for a push-pull configuration that should have lower harmonic distortion.
Here is an example relationship for QPA2237
[image]

gkasprow, 2021-03-19:
So, we can make a special version of Booster, dedicated to high-linearity applications.
If we go for let's say QPA2237, we would be able to significantly reduce the harmonics. With 360mA of Id, we can try to beat the ZHL-2W. Note that my measurements were at 22dBm, but harmonics were in fact 20dB higher due to the 20dBm attenuator.
ZHL-2W has -23dBm (after correction by 20dB due attenuator) of 2-nd harmonics at 22dBm output power.

gkasprow, 2021-03-19:
RFPD3580 is another interesting option, we would avoid connecting two stages in parallel

gkasprow, 2021-03-19:
I ordered a few push-pull wideband amplifiers, will give them a try.

hartytp, 2021-03-19:
@gkasprow thanks!

> So there is an obvious tradeoff: harmonics vs Idd.

If that's the situation here then I agree there isn't a design issue so much as a deliberate choice made. We wanted something with high power consumption for cases where linearity isn't particularly important. As you say, there is a reason this dissipates much less power than an equivalent MCL amplifier; we pay a price in terms of linearity.

This should be clearly communicated to users of Booster. If people want something with better linearity then we can look at designing a replacement RF circuit. First though it would be really helpful to have a clear spec for what is actually wanted from the design (bandwidth, max power, etc). I suspect that taking the current design but aiming for more linearity won't actually provide the optimal solution for people

hartytp, 2021-03-19:
> I cannot set a higher current than 250mA.

Why can't you go higher than 250mA? What limits this?

gkasprow, 2021-03-19:
Right. We can make an assembly variant with NPA1007 which has a similar package and pinout (only bias enters separate pin).
In this way the user can choose high efficiency of low distortion mode by just changing the Id.

hartytp, 2021-03-19:
what's the downside of using NPA1007 as standard and just using a lower current by default?

gkasprow, 2021-03-19:
250mA seems to the max due to thermal dissipation. I damaged the FET after running with a higher current. We are also limited by Booster 28V LDO - it trips.

hartytp, 2021-03-19:
> 250mA seems to the max due to thermal dissipation. I damaged the FET after running with a higher current.

Okay, so 7W quiescent max.

> I damaged the FET after running with a higher current. We are also limited by Booster 28V LDO - it trips.

Okay, good to know. IIRC this was a design choice by us in setting the foldback limiting resistors; the LDO itself can supply a lot more, but I may be wrong...

gkasprow, 2021-03-19:
>Okay, good to know. IIRC this was a design choice by us in setting the foldback limiting resistors; the LDO itself can supply a lot more, but I may be wrong...

exactly.

gkasprow, 2021-03-23:
I tried the [CATV amplifier](https://pdf1.alldatasheet.com/datasheet-pdf/view/16411/PHILIPS/BGD906.html). It works at 28V and provides -40.3dBc of second harmonics and -47dBc of third harmonics at 30dBm of output power.
At 20dBm of output power the second harmonics is -45dBc and the third one is at -60dBc.
The amp is a popular 8$ CATV hybrid. It's no longer produced but I could buy it quickly
It consumes 530mA at all power levels from 0 to 30dBm. 
It does not need any biasing. Just 28V. 
I ordered another one with similar specs and the same package from Mouser but these days it takes 2 weeks until they ship it...
Anyway, it looks promising. Such amp would use the cooling potential of Booster mechanics :)

gkasprow, 2021-03-23:
It seems feasible to make such an amp compatible with existing mechanics.

gkasprow, 2021-03-24:
some photos
[image]
[image]
power stage S11
[image]
power stage S21
[image]

[13 more replies]

closed 2022-02-07 by gkasprow