sinara-hw/Booster #384: Step Response Ringing

ibloomfield, 2021-06-16:
Hello-

We are running a booster and are experiencing ringing in a step response.

The booster is driving an AOM with an urukul -> booster -> aom configuration, f0=80MHz. Looking at the optical signal with a fast photodiode, there is ringing in the step trace. Looking at the rf out of the booster shows ringing and asymmetry in the waveform. The rf out of the urukul looks ok. Swapping in a Mini-Circuits ZHL-03-5WF+ in place of the booster removes any observed ringing.

To attempt to reduce this ringing, I tried calibrating the bias current to 50mA (and other values) with `biascal`, and also tweaking the DAC bias voltage with `bias`, neither seemed to make much difference.

Is this behavior to be expected? Might there be a way to reduce this ringing effect? Any help or input would be greatly appreciated - thanks in advance.

**AOM optical pulse:**
[image]

**RF pulse out of urukul:**
[image]

**RF pulse out of booster:** (note vertical asymmetry and ringing)
[image]

**Spectrum of booster** (apologies for the poor image quality, this is 10db per division, showing a 2nd harmonic of -9dBc. I did find high harmonic content being discussed in [issue 380](https://github.com/sinara-hw/Booster/issues/380)):
[image]

hartytp, 2021-06-16:
@ibloomfield is this problematic for you? What are your requirements?

Not sure how much others have played with this, but from a quick glance this all looks fit for the purposes that we originally designed Booster for (high power efficiency for driving Doppler cooling AOMs etc). So I can't say I have done any detailed investigation about this kind of transient.

ibloomfield, 2021-06-16:
The short answer is it is not clear yet whether this is problematic. Our goal is to be limited only by the acoustics, which in our optical configuration corresponds approximately to a 25ns rise time. The intensity fluctuations/ringing could cause issues but that is unknown at this time.

Could increasing the bias current with `biascal` potentially help this?

dnadlinger, 2021-06-17:
That ringing seems much larger on the optical trace than the RF trace. If you have a ~5 dB power attenuator handy, it might be interesting to see if the problem disappears when putting it at the Booster output to improve the S22, i.e. whether the issue is somehow related to matching.

jordens, 2021-06-17:
"square law detector"

ibloomfield, 2021-06-18:
### Comments
After some messing around with both `biascal `and the bias DAC settings yesterday, the system is looking better from the perspective of harmonics and ringing - see data below.

We weren't sure how the system would perform long term with a biascal of 250mA, so we've been monitoring for 16hrs. The temperature (meas:temp?) has been steady at ~28, and tx power (meas:out?) steady at around 34.8.

The biascal was failing at levels above 250mA. I tried lower DAC values with `bias 0 0` etc, and this helped reduce harmonics even more than biascal=250mA, but seemed unnecessary for long term operation.

I wasn't able to figure out how to manually write a biascal setting. I tried writing to the eeprom at addresses 28 and 29, but the output spectrum upon a power cycle did not look the same as when using the `bias` command. Does the biascal set anything in addition to the bias dac value?

Any comments on running at this bias current permanently? Concerns about thermal or anything else? Thanks!

 
### Data
**Scope traces** - setup is urukul->booster->directional coupler->AOM(tx port), scope on coupled port. aom signal to detector
50mA bias current, (purple = rf, blue=rf^2, green = optical)
[image]
 
250mA bias current
[image]
 
 
**Spectrum**
50mA
2nd harmonic -10dBc
[image]

 
250mA
2nd harmonic -24dBc
[image]

gkasprow, 2021-06-18:
I manage to kill the output stage by running it at 250mA. It dissipates 7W of power. So all depend on individual power stage thermal resistance to the heatsink. You can improve it by adding a thermal compound below the bottom RF PCB, under the power stage. It was not designed to work at such a high current so you are doing it at your own risk.
If you need low distortion, I'm just testing Booster_HL design which has far better linearity but takes a lot more power. Here is entire story https://github.com/sinara-hw/Booster/issues/380

ibloomfield, 2021-06-18:
@gkasprow Saw that- would potentially be interested in that hardware. For now we'll be running at 250mA biascal and see what happens.

Thanks for the thermal compound tip!

closed 2021-06-21 by ibloomfield