sinara-hw/Booster #387: TCP connection refused

margieb, 2021-10-02:
I'm trying to the SCPI interface for Booster and can't get a TCP connection going. I am using an IP address I gave the Booster (using the write command in VCP interfacing) on a local subnet (which I can ping and power cycle Booster to confirm it has the IP I think it does).
I first tried using TCP Client, a free app on the Microsoft Store, with this IP and port 5000, and I get the error:
"[E] Connect exception: No connection could be made because the target machine actively refused it"

I also looked through some relevant resolved issues on here and saw a python code involving:
>>> sock = socket.socket(socket.AF_INET, socket.SOCK_STREAM)
>>> sock.connect((<ip address>, 5000)
which gives me a similar error:
"ConnectionRefusedError: [WinError 10061] No connection could be made because the target machine actively refused it"

margieb, 2021-10-07:
@jordens ?

margieb, 2021-10-14:
This is a brand new Booster so I wasn't expecting it to have legacy firmware. Is there some other way to communicate with it?

dnadlinger, 2021-10-14:
Is there anything of note in the `logstash` output on the VCP?

By legacy firmware, Robert means the open-source one which predates the shiny new Rust-based one that QUARTIQ are selling separately.

margieb, 2021-10-18:
Is there documentation for that specifically for the booster? I see https://github.com/quartiq/miniconf and /quartiq/minimq for establishing a connection using Rust