sinara-hw/Booster #388: Inconsistent ADS7924 P28V0_CH readback per bias tuning

HarryMakes, 2021-12-07:
By running the bias tuning command, Booster will report different values of Ids (i.e. "the most recent P28V rail current measurements" by ADS7924) for the same given Vgs. Note that these RF channels are powered up for at least 2 minutes before their bias are tuned, which should give enough time for stabilisation for each power-cycle.

I wonder if the bias current has actually not stabilised before tuning can be done, or if the ADS7924 ADC cannot update the voltage register properly.

### Examples

On a channel 5, (Vgs, Ids) readback varies in 3 consecutive tuning runs.
```
Vgs = -0.740 V, Ids = 50.11 mA
Vgs = -0.740 V, Ids = 75.06 mA
Vgs = -0.740 V, Ids = 76.82 mA
```

On a channel 6, (Vgs, Ids) readback varies in 2 consecutive tuning runs.
```
Vgs = -0.680 V, Ids = 51.47 mA
Vgs = -0.680 V, Ids = 76.04 mA
```

On a channel 7, (Vgs, Ids) readback varies in 2 consecutive tuning runs.
```
Vgs = -1.080 V, Ids = 53.23 mA
Vgs = -1.080 V, Ids = 77.01 mA
```

jordens, 2021-12-07:
Looks related to #368

HarryMakes, 2021-12-08:
Thanks @jordens. We are also looking at increasing the bias current above 50 mA to reduce output distortion.

I just retried tuning the bias multiple times in a row, and it looks like this ~20 mA discrepancy only happen on the first (Vgs, Ids) readback via the `bias` command. This issue seems to happen only once every time the RF channel transitions from `Disabled` to `Powerup`, and regardless of the duration from the moment the state becomes `Powerup` to the moment the bias current is read.

I would've assumed that it's only that the firmware doesn't report the correct bias current only on the first read after the RF changes its state to `Powerup`. However, during bias tuning, multiple reads from ADS7924 are occuring for each bias voltage applied, so it's unlikely the firmware simply reads a delayed ADC value.

gkasprow, 2021-12-08:
@HarryMakes The ADC has an external lowpass filter after the mux. so it may be affected by the setting time of the mux  and the filter. I assume you don't change the mux during next ADC readouts

gkasprow, 2022-02-07:
did additional delay solve the issue?

jordens, 2022-02-08:
We haven't continued debugging the hardware. I would first try to resolve #368 and then revisit this here. But it should be looked at by an RF amplifier expert.

gkasprow, 2022-02-08:
I didn't observe such issue with old firmware.

jordens, 2022-02-08:
I don't think the legacy firmware does anything like this. You would not see it.

gkasprow, 2022-02-08:
That could be related to the AVDD issue. I replaced this resistor in new release