sinara-hw/Booster #395: Liquid cooling

jordens, 2023-09-26:
Let's evaluate designing a liquid cooling interface as an alternative to the fins and fans.
@gkasprow as discussed 

We had a mechanics repo but I think that's dormant/unused.

gkasprow, 2023-09-27:
I see 2 options. Modify the original design and instead of fins drill holes and install plugs. I did such things in the past[image]

It's optimal from thermal point of view

 Second option is to attach COTS liquid cold plates used in PCs to cool chipsets, RAM, SSDs etc. They are cheap but the design changes often with chipset evolution

I'd like to do it in elegant way using quick disconnect no drip connectors from Koolance. They are affordable and good.
One could replace the modules witout emptying the circuit.

gkasprow, 2023-09-27:
Another issue is how we circulate the liquid.
I use COTS radiatots with pump, resevoir, sensors.
It was sth like 200EUR when i bought it in 2015.
Then how we wire the circuits?
Do we connect them in series or in parallel?

Yet another option is big cold plate installed at the bottom of the enclosure and we place all modules on top of it. It could be problematic to make it and maintain optimal thermal resistances.

jordens, 2023-09-27:
Let's keep the internal and external chiller stuff separate. The external stuff doesn't appear to be so hard to supply and will also be application specific.
A common plate for the modules might also be problematic due to multiple existing mechanical constraints (connectors, existing screws). The gradient and crosstalk of series operation (or even parallel with a common plate) might be acceptable. We should first get a number on the power and phase tempos and time constat though.

gkasprow, 2023-10-12:
This is the proposed design
[image]
[image]

we will swap IEC power and cooling connectors so the latter are below the IEC. External connectors are non-spilt, so no need of draining liquid when changing connectivity.
[image]

The connectivity (series/parallel/mixed) may change.

gkasprow, 2024-07-18:
update: Booster cold plate set
[image]
[image]